PhiBabin/Redox-Lipo-Adapter #1: QUASO to integrate.

azhizhinov, 2023-05-07:
Hello.
I've remade the Redox keyboard with modern footprints.
We can add the QUASO at this PCB, as I can see, so I can share my KiCAD project files with you, if you interested in that.
And may I ask you about possibility to replace of BQ24075TRGTR with something more comfortable to solder with iron.
Thanks for your time. 
[image]

PhiBabin, 2023-05-15:
Yeah, the BQ24075TRGTR is kind of a pain to solder... Since your board seems to support the Pro-micro layout, probably the best way to get both wireless and LiPo battery is to use a Nice!Nano.

This wiki page is really good overview of all the battery charger chips, most of the chips in there are easier to soldier:
https://github.com/joric/nrfmicro/wiki/Batteries

azhizhinov, 2023-05-15:
Ok, thanks.

closed 2023-05-15 by azhizhinov