PhiBabin/Redox-Lipo-Adapter #2: Battery not charging

jadewolf05, 2023-06-25:
I made 2 of these(v1.2) and both have the same issue with the battery not charging.

When plugged in:
both have the charge led on.
5.1v between tp1 and gnd
battery voltage is present between gnd and both outside positive terminals that go to the redox when usb is unplugged.
battery voltage is present between gnd and tp1 when unplugged.
Battery voltage never increases(3.890 on one and 3.896 on the other)
USB ammeter also shows no current draw when plugged in.

Please let me know if anyone else is seeing this issue with V1.2 or if any other data is needed to determine why it isn't charging.

so far I have not installed them into my redox since they will not charge.

jadewolf05, 2023-06-25:
I've also noticed that no gnd is present on the gnd side of R2,R1,R4, and R5.  I also don't see a connection in the gerber files.  I have tried grounding the gnd side of those resistors, but no change in behavior.

jadewolf05, 2023-06-25:
missing gnd on R1,R2,R4, and R5 didn't sit right with me so I pulled up the datasheet for the BQ24075 and found that ISET(pin 16) is connected to R5 and per the datasheet, if ISET is disconnected charging is disabled. 
I tried grounding those resistors before, but after it was already connected.
I ran a jumper wire this morning to give those resistors ground on startup and BAM... now its charging. 

Hope this helps someone else running into this issue with V1.2.

jadewolf05, 2023-06-25:
For anyone reading this with the same issue.  
TLDR:
V1.2 doesn't have a GND connection on R1,R2,R4, and R5.
Run a jumper wire to give it ground or it will not charge.

closed 2023-06-25 by jadewolf05

PhiBabin, 2023-08-16:
Sorry for late reply.
I'm really sorry for that stupid mistake. Should have triple checked v1.2 before uploading it. I pushed v1.3 which is the same as v1.2, but with the missing via.
Sorry again for the troubles.