PhiBabin/Redox-Lipo-Adapter #6: Yellow light issue

OrvilleQ, 2024-07-22:
The yellow LED only lights up once when the USB is connected, instead of staying consistently lit when the USB is plugged in. This happens whether the battery is soldered on or not.

Is this the normal behavior of this board?

PhiBabin, 2024-07-27:
On both of my board it stay light up when the USB is connected (they have a battery connected, but the batteries are not fully charged). [Page 20 and 26 of the bq24075 datasheet](https://github.com/PhiBabin/Redox-Lipo-Adapter/blob/main/datasheets/bq24075.pdf) shows a diagram of state machine that control the CHG pin of the yellow light. Maybe your battery is just already charged?

PhiBabin, 2024-07-28:
I tried disconnecting the battery from the board and connecting the USB, and the yellow light only lights up once and then it shutdown.

OrvilleQ, 2024-07-29:
After re-soldering the BQ24075RGTR, the yellow light is always on when the USB is plugged in with the battery already connected. So this may be a different manifestation of the same problem as the battery not working, because you also reproduced that when there is no battery connected, the yellow light only comes on once, which seems to be normal behavior.

OrvilleQ, 2024-09-20:
Hi, I have another question.

After charging the battery for about 6 hours using the computer's USB port, the yellow light is still on, which makes me wonder if there is a way to identify if the battery is fully charged?

OrvilleQ, 2024-09-23:
Also the battery connected to the board can only been charged to 3.9V, 4.0V is the best. Is that by design? Sorry for this kind of basic questions, I only have limit knowledge about electricity, and it's my first time soldering a board myself. I tried to read the battery IC's document but cannot understand a bit of it :( .

Thanks for your help.

PhiBabin, 2024-09-28:
Interesting, BQ24075 should be charging to 4.2V. I don't know what could be causing this, like the [ battery IC's datasheet charge state flow diagram(p.26)](https://github.com/PhiBabin/Redox-Lipo-Adapter/blob/main/datasheets/bq24075.pdf) it seems like it'll keep charging until the charging current goes down to `I_term` (`I_term = 0.1 * I_chr = 10mA` ). I don't really understand how LiPo battery actually get charged, but maybe the charger just send current to the battery until either 4-6h (T_MAXCHG) has elapsed or if the current going in the battery as lower below `I_term`. If this is true, that would mean that the LiPo battery is not taking in any current once it reaches 4V or that it takes too much time to charge. I think it would be best if you just use your keyboard regularly and once it reach 3.3V or something, see if you can recharge it back to 4V.

I can try charging my keyboard for a few day and seeing the voltage I get. I vaguely remember getting 4.1V when I first build that keyboard, but that was a while back.

OrvilleQ, 2024-10-01:
Thank you for your reply. After communicating with the seller of the battery and observing and measuring the voltage of a keyboard that has been in normal use for a while, I currently suspect that the issue with charging the battery to 4.2V is related to the settings of the built-in protection board. I happen to be on vacation right now, and I will find time to disassemble the keyboard and test the charging current.

However, a new problem has arisen: besides my own keyboard, I also made one for a friend. His keyboard is unable to charge. After measuring, I found that the voltage of the two lithium batteries in his keyboard is 2.5V, indicating significant over-discharge. When I sent him the keyboard, I did not charge his batteries, so the only possibility I can think of is that the lithium battery module, due to being unable to enter sleep mode, continuously drew power from the nearly empty lithium batteries, ultimately leading to over-discharge. Is this possible? When the battery voltage is low, does this lithium battery module have a mechanism to protect the battery from over-discharge, and does it provide some visual indication (like a yellow light)?

Current issues:

1. The battery cannot be charged to 4.2V (I will test the charging current in the next few days).
2. Is there a visual indication for low battery and fully charged battery?
3. Is there a protection mechanism for low battery voltage?

I will provide further testing information as soon as possible. Please let me know if there are any tests you need me to conduct. Thank you.

PhiBabin, 2024-10-07:
The circuit has no under voltage protection, so it'll drain the battery if the keyboard is left on. This is a mistake on my part, I misunderstood the LDO worked and assumed that at <3V it would shutdown. I based the circuit on the Nice!Nano! v2 and [looking online](https://www.reddit.com/r/olkb/comments/1ajjl9m/can_i_use_these_batteries_with_my_nicenano_clones/kp3ab2f/) it seems that the circuit was designed with the expectation that the battery has its own undervoltage protection. I'm really sorry about that.

I'll look into and see if I can find an easy fix, but it seems that I'll probably have to redesign the board with a different Battery Manager chip (one with undervoltage protection) or with an additional under voltage protection circuitry.

OrvilleQ, 2024-10-08:
@PhiBabin Thank you for your response. I don't believe using batteries with protection circuits constitutes a design flaw on your part, because most of the soft-pack lithium batteries available today, at least from my perspective, come with protection circuits. In fact, the two batteries that were over-discharged also had protection circuits, and their cutoff voltage for discharge should have been 2.75V. I don't see this as a design defect that requires fixing—perhaps just adding a note in the documentation would suffice.

As for my progress, the new lithium batteries I ordered from another vendor have finally arrived. Upon receiving the batteries, I measured the voltage, which was 3.982V.

[image]

I connected the lithium battery and multimeter in series with the circuit on the lithium battery board and measured the idle current consumption of the battery board to be 20uA， and when a button is pressed, the current consumption is approximately 10mA.

[image]

Finally, I plugged the charging port into the circuit board. When the yellow light turned on, indicating the start of charging, the circuit board was still consuming around 17uA of current. It seems that the current is not being delivered to the battery at all.

[image]

Since the battery I purchased supports 1C charging, I also tried shorting JP1 for the experiment, but the situation remained basically the same as when it was not shorted.

OrvilleQ, 2024-10-09:
The board pass most of the test you mentioned in BUILD.md, including:

- Plug the USB-C port and check the voltages you should see:
  - 5V between test point TP1 and the ground
  - 3V between the holes on either side of the board and the ground.

- Check that when the USB is plugged in, the LED lights up.
  - Check that when the USB is not plugged in, the battery is providing power:
  - the voltage between BAT+ and ground should be the same as between test point TP1 and the ground;
  - And you should also see 3V between the holes on either side of the board and the ground;

The only test I didn’t follow strictly was the charging test. The steps I followed at the time were:

1. After soldering the battery, I turned on the keyboard switch but found the keyboard wasn't working.
2. I plugged in the USB-C port and observed the yellow light turning on. After a while, I unplugged it and found the keyboard was working fine.
3. I felt everything was okay and assumed the charging was successful, though I didn't measure the voltage during the process.

Just bought a tool for discharging the battery. I plan to discharge the battery to around 3V and then recharge it to see if it can charge properly.

PhiBabin, 2024-10-17:
> Thank you for your response. I don't believe using batteries with protection circuits constitutes a design flaw on your part, because most of the soft-pack lithium batteries available today, at [...]

I'll update the README.

I let my keyboard charge for a week and it got charged to 4.1V,  which match the voltage I was getting when I first build the board.

> 1. After soldering the battery, I turned on the keyboard switch but found the keyboard wasn't working.
> 2. I plugged in the USB-C port and observed the yellow light turning on. After a while, I unplugged it and found the keyboard was working fine. (the charge could be less than 20 seconds on some board)
> 3. I felt everything was okay and assumed the charging was successful, though I didn't measure the voltage during the process.

Thanks for taking the time to test your battery! So you're getting only 20 seconds of charge on a ~3.9V battery? It sounds like there is something not properly soldered, for instance, if the ISET pin is not properly soldered, charging is disabled.
Just to confirm, you're on board v1.3? Because V1.2 has an issue where this pin was not routed properly (R1, R2, R4 and R5 weren't connected to GND). FYI, I'm on board V1.0, so it's totally possible that I made another mistake on v1.3.

Looking at the gerber file for V1.3, I noticed that the via for R2 might be too far away from the pad. Could you check the continuity between the  CC1 pin of the USB and R2? In the following screenshot I have put an arrow on the via that might be too far from the pad:
[image]

Note: CC1 is only relevant for USB-C to USB-C, for USB-A to USB-C, it should have no impact.